VoronDesign/Voron-0 #38: Gaps in Toolhead Blower Duct (Left)

mvrckcompany, 2020-09-05:
Thin walls around the screw holes on the part: "[a]_Toolhead_Blower_Duct_Left_TV6_x1" leads to gaps on the inside during slicing and printing. Attempted to alleviate the issue by enabling "Print Thin Walls" within Cura, to no avail. 

Printed using recommended settings, please see attached for reference.

[image]

[image]

[image]

Roycinger, 2020-10-04:
I can confirm that S3D behaves in a similar fashion:
[image]

I'll have a try with SuperSlicer tomorrow

FHeilmann, 2020-10-19:
I checked back with the devs and they assured me that these holes do not impede the functionality of that part. Yes they are a little unsightly, but unfortunately the space around these featres does not permit a "fix" via thickening those sections.

I'm therefore going to close this issue. If you feel this has been in error and want to provide a rationale while this should be fixed apart from the visual aspect, feel free to reopen it!

Bests,
-Florian

closed 2020-10-19 by FHeilmann

FHeilmann, 2020-10-25:
@mvrckcompany the devs elected to fix your issue anyway. Latest master (with https://github.com/VoronDesign/Voron-0/pull/45 merged) should fix the slicing issue.

commit 2eb4463e (2020-10-25): Close holes in blower duct, fixes #38 (#45)

* Delete [a]_Toolhead_Blower_Duct_Left_TV6_x1.STL

* Add files via upload

fix print slicing from issue #38