VoronDesign/Voron-0 #43: Deck panel rear DXF missing segments

alatdneg, 2020-10-04:
I'm loading it into Freecad 0.18.4, and there are two curves missing, not sure if this is a FreeCAD bug or not, but the middle back plate loaded fine.

It's the two curves that connect the z-carriage inset.  --\___/-- 

ta

Roycinger, 2020-10-04:
[image]
[image]
[image]

No issue for me here (QCad, Illustrator, Inkscape). Must be FreeCAD related

alatdneg, 2020-10-05:
Yup, FreeCAD seems to be a bit iffy on DXF's, I loaded it into Inkscape and re-exported as SVG, that loaded okay into FreeCAD.

I guess it might be nice if you included STLS of the plates at 3mm, 2.5mm  etc. As I imagine quite a few people would like to just print these, and it'd save them having to create the STL's from the DXF's which might be difficult for some people.

closed 2020-10-05 by alatdneg