VoronDesign/Voron-0 #57: Setup Guide Startup Pin No Serial Device

jgehrig, 2020-12-30:
The [Setup Guide](https://github.com/VoronDesign/Voron-0/blob/master/VORON-0/Manuals/V0_Setup_Guide.pdf) mentions setting `!PC13` on startup in the config. On my device with `!PC13` the serial port does not appear in `/dev` and no device is listed in `lsusb` either.

I resolved the issue above  by changing `!PC13` to `!PA14`. I suspect there was a hardware change in revision `V2.0`?

I'm not familiar with Klipper/SKR, and I can't find a good schematic to validate... Does this sound reasonable?

Upstream reference:
`https://github.com/KevinOConnor/klipper/issues/3461`

dubjay2k, 2020-12-30:
@jgehrig The SKR mini V2 does use a different pin than what's outlined in the setup guide (v1.2 specific).  

It's mentioned in the V2 `printer.cfg` file, but should probably be mentioned in the setup guide as well.

https://github.com/VoronDesign/Voron-0/blob/74603b1752b24fcdebdb85c9d662b88075058e6f/VORON-0/Firmware/SKR_E3_Mini_2.0/printer.cfg#L1-L5

raymondh2, 2021-02-12:
Closing due to inactivity

closed 2021-02-12 by raymondh2