VoronDesign/Voron-0 #58: Adjustment suggestions for the manual

Metti93, 2021-01-02:
Hello all
First of all I would like to say that I, a CAD designer, am very enthusiastic about this great printer. The parts, as well as the manuals and the community are of high quality and unparalleled.
That's why I want to use my skills as a quality manager and responsible for company standards in my company to make this great project even better. Starting with the manual!
When I assembled my V0, I felt a bit lost in certain parts of the manual.

**General hint:**
- That Loctite might react with ABS+ and might not be beneficial in certain places. A note on this might not be so bad, for noobs like me.
- I would make the text bubbles and their text larger. I'm not the youngest anymore, and would have liked more readable texts. Also, I'm not a fan of white text on an orange background.
- I would introduce a bubble/color for optional hints. Be it additional M3 nuts for future extensions etc..

**Page 6-7**
- I would number the profiles. Because I made the mistake of installing the wrong profile right at the beginning, which is what should be needed with the frame. Of course, I then simply added two holes. But that could have been avoided.

**Page 10-11**
- "2x" is not in the same place, on both sides!!!!! 😉

**Page 15**
- Bottom left illustration. Write on, nevertheless, please, that one puts a M3 nut under there in each case. I had wondered what it is held by, or if on any other pages this screw is used for something else, because it sits there loosely.

**Page 15-16**
- As an optional note, I would add additional M3 nuts lined with foam tape (thanks Christoph Müller#3576 for the tip).
Anyway, I would be happy if I had received such a tip early enough, and not only when I have assembled everything.

**Page 20**
- The title of the page is not aligned correctly.

**Page 25**
- A hint as to where the motor cable should look would be nice.

**Page 27**
- Backlash nut should be written on. Only at the end of the assembly, did I notice that this was missing. This was probably the worst thing missing from the manual.

**Page 29**
- Illustration at top left, "M3*6 BHCS + M3 Groove." Not that people use MakerBeam T-Nuts.

**Page 30**
- Foam tape would hold the nuts in place here, making it easier to mount the Z-motormount, but also the backpanel.

**Page 34**
- "M3*6 BHCS + M3 Nut."

**Page 35-36**
- M3 nuts, secured with foam tape, for future upgrades.

**Page 37**
- A measurement with the edge distance would be helpful, not that you tighten the rail in the completely wrong place already.
- Text "Use Rail Guides to Center Rail" is missing.

**Page 74**
- The M3 nuts on both sides of the rail should be listed, noticed they were missing on page 81. Had to take everything apart again.

**Page 136-137**
- The cable routing for the front fan, I would draw in better and route it through the notch for the left part fan.

For questions I am also available in Discord
All in all, it was fun to assemble the Voron.

mysfiring, 2021-01-15:
Didn't want to make a separate issue for a manual-related item: a callout on pg. 10 of the Pocketwatch manual - as a reminder to those using a non-spec motor - reminding the user that they should attach the 'pw_motor_spacer' at this point would be greatly appreciated! :D

nemgrea, 2021-02-20:
manual is getting re-made for v0.1 these issues will be taken into consideration and implemented where applicable

closed 2021-02-20 by nemgrea