VoronDesign/Voron-0 #61: FR : More FDM friendly tophat parts

ChipCE, 2021-01-26:
Hello, sorry if this is not the right place to request FR like this.
The chamfer on the top hat parts is 36.78° that make the parts very hard to print nicely with FDM printer. At first, I was thinking problem is my bad cooling, but other members on Voron discord also have the same problem, some suggest to print the parts with PLA+. If the chamfer angle is 45° or higher, the parts will can be printed with most of filament type.
[image]

Stephan3, 2021-01-30:
You can print the 4 piece tophat in a 45° orientation thats more fanfriendly. That helps a lot

ChipCE, 2021-01-30:
Yes, I printed with the 4 pieces version. The picture above is cad model of the 4 pieces version. 

[image]
[image]

nemgrea, 2021-02-20:
this is being fixed in the next release, closing the issue for now as the new tophat will eliminate the <45deg faces

closed 2021-02-20 by nemgrea