VoronDesign/Voron-0 #77: BOM discrepancy?

bogeyf14, 2021-04-25:
BOM calls for "Bondtech Motor Gear" but it is unnecessary because the spec'd stepper already has a gear

VinnyCordeiro, 2021-04-25:
It is still necessary for those who opted for the bowden setup.

bogeyf14, 2021-04-25:
ok, I would recommend appending the description of that BOM line to say " Bondtech Motor Gear (required for bowden extruder)" or something to that effect. The manual doesn't show assembly of a bowden setup so it's not clear what this part is used for. I am sourcing parts for for the new toolhead and almost ordered this part because I didn't check the manual first, I doubt many other people will either.

raymondh2, 2021-04-25:
You are looking at two different versions of Voron zero

closed 2021-04-25 by raymondh2

nemgrea, 2021-04-25:
issue fixed thanks for letting us know!