VoronDesign/Voron-0 #82: Assembly Manual is corrupt

lococnc, 2021-05-02:
Manual updated 2 hours ago won't open. Adobe reports the file is corrupt.
Tried downloading twice, deleting file in-between.

ragraham91, 2021-05-03:
I cannot replicate.

Steps taken:
Visited https://vorondesign.com/voron0.1 and clicked 'Manual' (https://github.com/VoronDesign/Voron-0/raw/Voron0.1/Manuals/Assembly_Manual_0.1.pdf)

Result:
Assembly_Manual_0.1.pdf downloaded and opened without problems using Adobe Acrobat Reader DC.

nemgrea, 2021-05-05:
im also unable to replicate the issue

steps:
visit https://github.com/VoronDesign/Voron-0/blob/Voron0.1/Manuals/Assembly_Manual_0.1.pdf
use download button to download. 
opens fine

im closing this issue until more details are provided

closed 2021-05-05 by nemgrea