VoronDesign/Voron-0 #89: Recommend revisions to V0.1 bed construction

fwiffo, 2021-05-15:
Having gone through a couple rebuilds of my bed, I'd like to offer a few suggestions for refinement:
1. Ditch the printed thumb knobs and go back to the knurled metal nuts from V0.0.
2. The Z adjustment screw hole strips out quickly. Thicker walls or something might help.
3. Put washers between the springs and lock nuts to keep them from getting cockeyed and buckling. I had a bad problem with mine shifting around on the nuts while adjusting. Adding a washer gave them a flat surface to push against.
4. Make the holes for the springs a bit shallower. That gives them a bit more tension at a given height making adjustments more reliable and gives a wider usable range.

nemgrea, 2021-05-20:
spec will stay with printed knobs for bom/cost reduction purposes. metal thumb nuts are still supported if users choose that route

im not sure what z adjustment holes would be stripping? we dont have any bolts threading into printed parts on z axis stuff

looking into adding the washers to help with the flat surface issue. this will require cad/manual updates

the hole depth for the springs is at its current depth to allow the full 120mm travel in z having the bed sit higher up may eat into this space. simply tightening the thumb nuts should compress the screw to whatever preload amount you would want. these parts will stay as modeled.

fwiffo, 2021-05-20:
I'm talking about the screw that hits the Z endstop switch (page 102 of the
manual).

Megamannen, 2022-02-07:
Threading in plastic is really bad if you adjust it often. But I'm guessing this is a "set once or twice" for most users, in that regard it doesn't matter.

But the Voron is filled with threaded inserts, so what is one more?

staffert, 2022-02-09:
Simply using a second locknut would provide ample locking of the screw, or a brass heat insert with thread locking compound would be a better long term solution than simply a hole into plastic.

closed 2022-08-04 by nemgrea