VoronDesign/Voron-0 #97: Is it possible to scale to 4040 extrusion?

hjf, 2021-05-30:
1515 extrusion is not available worldwide. There is no equivalent of McMaster-Carr or Misumi outside the USA. It's extremely expensive to have shipped internationally too.

4040 extrusion is metric, easy to find pretty much anywhere, and less than 2mm larger.

Is it possible to scale the printed parts to account for this?

Also I hope if there is a 1.0 (or 2.0) version, this is taken into account and revised for the more standard size.

raymondh2, 2021-05-30:
No there is no way to just scale the parts to work with with 4040. Also 1515 is 15mm x 15mm

hjf, 2021-05-30:
> Also 1515 is 15mm x 15mm

That's strange. Googling for 1515 extrusion always leads to 1.5x1.5". I thought it may be 15x15mm (since 2020 is 20x20mm) but almost all results I found are imperial sized.

Well, the second part of my comment still stands. In that case, I still suggest keeping this in mind for future revisions. 2020 in this case, which is actually even easier to find than 4040

hjf, 2021-05-30:
Never mind, someone already did: https://github.com/gsl12/Tiny-M

closed 2021-05-30 by hjf

raymondh2, 2021-05-30:
We have no plans to change the extrusion profile on V0

VinnyCordeiro, 2021-05-31:
@hjf You can buy the MakerBeam XL profiles, the original V0 design was intended to use them. They are available pre-cut and pre-threaded, and can be shipped worldwide, although I have to agree about the shipping costs here.