VoronDesign/Voron-0 #103: Zip ties missing from BOM

grmocg, 2021-06-09:
Assembly requires zip ties in various locations, but they're not included in the BOM.

... And since some of the parts require smaller zip-ties than people are likely to have (smaller than 4in/100mm, which are typically the smallest in normal zip-tie assortments), it is worth a call-out somewhere like the BOM/sourcing guide!

nemgrea, 2021-06-11:
zip ties added to sourcing guide, thanks

samtny, 2021-06-21:
FWIW upon encountering this issue I grabbed these: https://www.amazon.com/dp/B0826Z1RLK

The quality is ... okay (minor burrs here and there), and my caliper says they are indeed 1.8mm width as described for the product.

tawera, 2021-07-08:
A suggestion for improvement: making the zip tie mount points compatible with 2.5mm would make sourcing easier. At least where I am from, I've checked many Swedish suppliers, all have 2.5mm, none have 1.8mm. And, incidentally, the ones that came with the prusa kit I bought were 2.5.

tme.eu has 1233 SKUs for zip ties, only 3 of which are 1.8mm or less, so these are rare...

Megamannen, 2022-01-15:
> A suggestion for improvement: making the zip tie mount points compatible with 2.5mm would make sourcing easier. At least where I am from, I've checked many Swedish suppliers, all have 2.5mm, none [...]

Yeah, 1.8 is indeed a rare thing, here you have one (the only?) supplier though https://www.electrokit.com/produkt/buntband-71mm-x-1-8mm-svarta-100-st-panduit/

raymondh2, 2022-04-01:
Added to the BOM

closed 2022-04-01 by raymondh2