VoronDesign/Voron-0 #111: Firmware/printer.cfg needs run current changes, too high for OMC Z-axis motors

KurioHonoo, 2021-06-17:
```
[tmc2209 stepper_z]
uart_pin: PC11
tx_pin: PC10
uart_address: 1
interpolate: True
run_current: 0.37       # For V0.1 spec NEMA17 w/ integrated lead screw   
hold_current: 0.35
sense_resistor: 0.110
stealthchop_threshold: 500
```
The `run_current: 037` is fine for the LDO-42STH25-1004CL200E 1.0A, but is over the 0.28A max for the OMC 17LS13-0404E-200G 0.4A

Perhaps having two different configs in `[tmc2209 stepper_z]`, with the OMC being commented out by default, but adding additional commenting to inform the user.

I've added an example below as well as included all of the current BOM motors and their currents in case that information was helpful or if you wanted to incorporate them as comments into the config. 

```
[tmc2209 stepper_z]
uart_pin: PC11
tx_pin: PC10
uart_address: 1
interpolate: True
#run_current: 0.37       # For V0.1 spec NEMA17 LDO-42STH25-1004CL200E 1.0A  
#hold_current: 0.35
run_current: 0.2         # For V0.1 spec NEMA17 OMC 17LS13-0404E-200G 0.4A 
hold_current: 0.15
sense_resistor: 0.110
stealthchop_threshold: 500
```

[tmc2209 stepper_x], [tmc2209 stepper_y]
OMC 14HS20-1504S 1.5A
LDO N/A

[tmc2209 stepper_z]
OMC 17LS13-0404E-200G 0.4A
LDO-42STH25-1004CL200E  1.0A

[tmc2209 extruder]
OMC 14HR07-1004VRN 1.0A
LDO-36STH17-1004AH 1.0A

hyperphonic0, 2021-06-19:
If you were going to have both configurations, it should probably the OMC motor that is uncommented by default because:
- The lower current setting of the OMC motor avoids any potential damage to the LDO motor, but the same is not necessarily true the other way around.
- The OMC motor is the _recommended_ motor in the sourcing guide, the LDO motor is the '_alternative source_'.

KurioHonoo, 2021-06-20:
I originally was going to say OMC too. Good point on it being the recommended one.

samtny, 2021-06-23:
Agree, choose the reasonably lower current as the default. My OMC was running super hot (2nd degree burn hot) before I caught this just now.