VoronDesign/Voron-0 #121: 30mm blower fans sensitive to the hotend heat and stops working

bmgoncu, 2021-08-22:
The BOM specified GDSTime 30mm blower fans used for the Part cooling fans are susceptible to overheating from being in close proximity to the hotend.

Observed behaviour;
- When the hotend is hot the fan on the side of where the heater cartidge tip is visible, ossilates and doesnt spin 
- When cooled down or when the hotend is not driven the fan spins as expected.
- The other part cooling fan spins normally.

Potential fixes;
- A not so temprature sensetive blower fan
- Somehow isolating the heat of the hotend from the fan hub.

Also asking for potential workarounds and/or solutions.

bmgoncu, 2021-08-23:
İ have done some further testing with a makeshift heat shield and it does appear to alleviate the problem. Havent tested with the chamber closed and heated but i think this is a good start;
https://youtube.com/shorts/RvsFIU7-kds?feature=share

https://youtu.be/qGwNJFiHI5k

bmgoncu, 2021-08-24:
Added the shielding with vhb to the fan (not ideal i know) the issue presides albeit the fan transitions to that state later (might be a placebo effect havent occured to measure the time).

bmgoncu, 2021-08-24:
More of the same issue i think;
https://discord.com/channels/460117602945990666/696930677161197640/877269535609212939

pfmooney, 2021-08-25:
Yep, I've seen this as well.  I even tried pushing the kickstart time up to 2s.  In a warm chamber, even 30% was too low for them to stay spinning sometimes, whereas 20% was fine cold.

pfmooney, 2021-08-28:
Al foil as a reflector does not appear to have helped at all.  I swapped the fan on one side for one of my spares and did not seem to help either.

lococnc, 2021-09-08:
Delta fans do not have this problem. My GDSTime fans lasted a day. Replaced and have no issues. Mine are 12v so it took adding a buck,

pfmooney, 2021-09-08:
> Delta fans do not have this problem. My GDSTime fans lasted a day. Replaced and have no issues. Mine are 12v so it took adding a buck,

That's for prints in a warm chamber, or in open air?

lococnc, 2021-09-08:
Sealed Chamber, printing ABS, I run fans 1t 16%

bmgoncu, 2021-09-08:
Can you provide an aliexpress link so i can verify? Maybe the BOM may change if it is indeed superior.

lococnc, 2021-09-08:
I bought from Digikey
https://www.digikey.com/en/products/detail/delta-electronics/BFB0312HA-A116/5022815
I used a small buck-converter to power the + terminal on fan, since the MCU switches ground.

blalor, 2021-09-08:
When talking fans, there’s little value in citing the brand or vendor as the sole detail. The DigiKey link is ideal. The useful info is in the specs, which are seldom provided on aliexpress, Amazon, etc. CFM, static pressure, voltage (obviously) and noise are all relevant.

lococnc, 2021-09-08:
Ali link to one identical to the ones I used.
https://www.aliexpress.com/item/1005003179503083.html?spm=a2g0o.productlist.0.0.74d52c3aBSuXD7&algo_pvid=c56dd7c2-b18e-4b14-8008-ffcd0abfa046&algo_exp_id=c56dd7c2-b18e-4b14-8008-ffcd0abfa046-1&pdp_ext_f=%7B%22sku_id%22%3A%2212000024528815771%22%7D

Hexide, 2021-09-12:
Update on my insulation attempt: 

After ~70 hours of printing, GDSTime fan started showing same failure symptoms. So issue was simply postponed rather than being fixed.

This calls for BOM fan to be changed to something that works better.

Personally, I will give a shot for listed Delta fan.

bmgoncu, 2021-09-12:
I already ordered a set, will arrive in 2 weeks i think. Will update with results.

xnaron, 2022-01-07:
@bmgoncu did the new fans work?  I noticed the Delta fan linked above is 12V.  Any issues with the airflow?  I guess it is better than a non spinning gdstime one :)

bmgoncu, 2022-01-07:
@xnaron yeah they totally do work, the airflow is signifocantly better (As it is the noise) and the best part no heat shielding at all (sorry for the radio scielence totally got carried away by other things and didnt report back)

xnaron, 2022-01-07:
@bmgoncu Awesome thanks!  Did you use a buck converter to get the 12V for them?

bmgoncu, 2022-01-07:
Actually no, i use the split wire config described in the doc

lococnc, 2022-01-08:
Buck it.

aschiweck, 2022-07-17:
> * Fan works fine when everything is cold
> * Fan maybe works for few layers if they are quick
> * Fan does not work by the end of print

I can confirm this issue also using a mosquito hotend.

I replaced the BOM specified GDSTime blower fans some time ago by ones from WINSINN, 30mm, Blower, 3010, Dual Ball Bearing, 24V. They are working smooth since then, regardless of heat or hotend state, no buck needed.

I bought them from [AmazonDE](https://www.amazon.de/gp/product/B08MKLDGQR/) but i think you can can get them also from https://winsinn.com/30mm-fan-raspberry-pi/ (3010 Blower - Dual Ball Bearing)

szafran81, 2022-12-10:
Both links from sourcing guide redirect to GDSTime fans (just checked).

Mstaniforth94, 2023-01-22:
I've recently been through about 6 of these fans, tried fan saver and this didn't help, and others on the Voron discord are still having issues with the fans on MiniSB.
https://discord.com/channels/460117602945990666/461210576618651669/1066792849221357578
Even with its new fan-saving design. 

I have had one GDstine that's done 500+ hours and another do 20 minutes, 2 hours, 5 hours, 50 hours... on fully heat-soaked prints. 

An inconsistent quality supplier is not a good quality one, with all the evidence and issues... I suggest these fans be removed from the BOM and the 5V Delta fans become the go-to. They may be double+ the price, but if that means you use less than double the fans over time, I think that's a better sourcing decision especially for newcomers just wanting to print stuff, and forcing the kit suppliers to send the better fans.

I did find cracking the door open stopped the issue on one of my GDstine fans, with fan saver too, but resulted in chamber temps cold enough to warp ASA on larger parts, thus meaning I was limited to ABS+ and small parts only.

szafran81, 2023-01-22:
My GDSTime did one Voron cube print, and after that they died each print after about 5mins.
Now I use the WINSINN 24V 3010 ones, and they work great.

porkcube, 2023-02-14:
Had GSTime 24V 3010 blowers on my 0.1, total failure of one in under 100hrs, replaced with Fabreeko upgrade kit and have almost 400hrs of them working fine since.

Separate new 0.2 build have been fighting the blowers, replaced right side 3 times before finally got a good cube out of ABS, then the left side started failing.   Tried fansaver and non-fansaver mounts which made no difference, only opening the door allows for a decent print with ABS.

Skystrike005, 2023-03-07:
I'm facing the same issue as well in miniSB. fan saver + dragonfly BMO lead to dead fans when the chamber is at printing temp. Fans are brand new, less than 1 hour of printing time.

Will try replacing the fans with winsinn ones. Heard they're much more stable. If all fails i'll try using dragon burner with 4010 fans.

nemgrea, 2023-03-16:
SG switched to winsinn fans mid feb. we'll try these for a bit. these are the fans i personally use in both v0's no failures yet.

danicedenv, 2023-03-20:
@nemgrea did you see any relevant dip in cooling performance? I ordered some Winsinn fans because my GDSTimes can't keep up with the heat. I am only hesitant because as far as I understand they don't push as much air and I mostly print PLA. Then again I don't want to give up being able to print ABS properly.