VoronDesign/Voron-0 #128: Incorrect Stealthchop Setting in the Default Config

hyperphonic0, 2021-10-18:
There is quite a lot of discussion on Discord right now, with people having problems with the wrong default setting in the repo.  This has lead to many turning off Stealthchop entirely, meaning they will not benefit from near-silent - _and skip free_ - steppers.

[Klipper docs](https://www.klipper3d.org/TMC_Drivers.html) are fairly clear that Stealthchop should either be always on, or always off:

> ### Enabling "Stealthchop" mode¶
> 
> By default, Klipper places the TMC drivers in "spreadcycle" mode. If the driver supports "stealthchop" then it can be enabled by adding stealthchop_threshold: 999999 to the TMC config section.
> 
> It is recommended to always use "spreadcycle" mode (by not specifying stealthchop_threshold) or to always use "stealthchop" mode (by setting stealthchop_threshold to 999999). Unfortunately, the drivers often produce poor and confusing results if the mode changes while the motor is at a non-zero velocity.

So I would suggest changing the default configuration for `stepper_x`, `stepper_y` and `stepper_z` to:
 
`stealthchop_threshold: 999999        # Comment out to turn off StealthChop`

I doubt that using Stealthchop on the extruder motor is going to provide any benefit, and because it can reduce the motor torque at higher speed it may cause problems. I would suggest turning it off for the extruder.

Stephan3, 2021-10-18:
https://github.com/Klipper3d/klipper/blob/5c10001bc525701f0ef004041cbbf093522f6de6/klippy/extras/tmc.py#L512

Take a look at the code. If value is 0, pwm mode is false and therefor the field in the register is too. 
With a value of 0 stealtchop is disabled.

hyperphonic0, 2021-10-18:
I'm not quite sure what you're saying, but I'm suggesting the default config should have Stealthchop turned on by default, with the included comment explaining how Stealthchop can be turned off, if that is what is desired.

I believe setting `stealthchop_threshold: 0` would also turn it off, as would excluding the setting from the config.

samtny, 2021-10-21:
lol, i put a PR in for the extruder one, didn't see this issue:

https://github.com/VoronDesign/Voron-0/pull/127

#127 

PR turns off stealthchop on the extruder only. I have personally troubleshot this with at least five people in the discord since I built my V0.1, July-ish.

closed 2021-10-31 by nemgrea

hyperphonic0, 2021-11-02:
@nemgrea not sure why this was closed, #127 doesn't resolve this issue, it turns Stealthchop off for the extruder but does nothing the A/B drives or Z axis.

I think my original suggestion is still correct. I can create a pull request if you’d prefer, but thought an initial discussion here would be a better point of reference; It can be a bit difficult to discuss this kind of issue on Discord!

PR #127 (2021-10-31): turn stealthchop off extruder
I have observed in Discord a bunch of folks with extruder skipping issues still running the default "500" `stealthchop_threshold` value which essentially sets the extruder stealthchop to "always on". Turning it off / setting to 0 has fixed issues for a couple folks. I believe this is de rigueur to leave it off on the pancake steppers in any event, afaik ymmv.

Partially resolve #128