VoronDesign/Voron-0 #137: PrusaSlicer Profile extrusion width

gsezz, 2021-12-09:
The extrusion width percentages in the PrusaSlicer profile seem to be based on Nozzle diameter, like SuperSlicer does. 
PrusaSlicer calculates them on Layer height.

raymondh2, 2021-12-18:
I'm not entirely sure what you are asking. Are you refering to the one build into Prusa Slicer? If so we don't maintain that.

gsezz, 2021-12-18:
No, I'm refering to the [VoronV0.ini](https://github.com/VoronDesign/Voron-0/blob/Voron0.1/Slicer_Profiles/PrusaSlicer/VoronV0.ini) here on github. For my Understanding, it's supposed to work with Prusa Slicer. But `extrusion_width = 110%` doesn't make much sense in Prusa Slicer, since it bases percentages on Layer height, not on Nozzle diameter, linke Super Slicer does. So while 110% sounds reasonable for Super Slicer (0.44mm extrusion width with a 0.4mm nozzle), this setting would lead to 0.22mm in Prusa Slicer (based on 0.2mm layer height).

nemgrea, 2022-01-05:
fixed

closed 2022-01-05 by nemgrea