VoronDesign/Voron-0 #139: 5V Hot End Fan in to 24V FAN0

SanderRave, 2021-12-29:
Hi team,
I crossed this discrepancy by accident while searching for configuration info in Voron Documentation under Voron V0.0 - SKR mini E3 V2.0 Wiring: https://docs.vorondesign.com/build/electrical/v0_miniE3_v20_wiring.html#voron-v00---skr-mini-e3-v20-wiring

Bulit 4; Plug Hot End Fan in to FAN0 (pin PC6), is FAN1 (pin PC7) in the example Github printer.cfg: https://github.com/VoronDesign/Voron-0/commit/fa6f00e1f4592a6e5fb8976414297ccbbb6d4b64
Both pins have 24V output.

However, in the Voron sourcing guide, a 3007 axial fan 5V (min 3.5CFM) is specked: https://vorondesign.com/sourcing_guide
Connecting the fan straight to FAN0 or FAN1 will fry your hot end fan. With a lack of +5V outputs, my positive is wired to the probe +5V pin and PIN1 ground. Not sure where that wisdom came from, but I didn't make it up by myself.

Hope this helps making future building experiences even more awesome!

alvius, 2021-12-30:
You are looking at v0.0 documentation and v0.1 BOM. Check v0.0 BOM or v0.1 documentation.

raymondh2, 2021-12-30:
Closing as the referenced docs are not for the correct revision of Voron Zero

closed 2021-12-30 by raymondh2