VoronDesign/Voron-0 #141: Panel_Mounting/For_3mm_Panels/ are incorrect thickness

Keithcuda, 2022-01-05:
Voron-0/STLs/Panel_Mounting/For_3mm_Panels/ are the same dimensions as for the 2.5mm.
Only the Rear bottom corner seem to be correct for 3mm panels.

Keithcuda, 2022-01-05:
Sorry, just spotted #112 has already been logged.
As these are the size that LDO kits supply, i am surprised it has not yet been corrected.

Keithcuda, 2022-01-05:
I'm wondering if the 3mm Panels are designed to not have the VHB tape? I've done this and seems ok. In which case the Manual will need to be updated to reflect this.

hyperphonic0, 2022-01-05:
Is this closed issue relevant: #125?

Keithcuda, 2022-01-05:
Yes, I think the Manual needs updating to instruct not to use the VHB tape on the sides and back, with the exception being for the Front Door panel as it is not fixed to the frame therefore requiring the tape.

atoomnetmarc, 2022-02-18:
> Yes, I think the Manual needs updating to instruct not to use the VHB tape on the sides and back, with the exception being for the Front Door panel as it is not fixed to the frame therefore requiring the tape.

Rear bottom seems to also need some VHB.