VoronDesign/Voron-0 #158: Mosquito_Toolhead_DD for 3007 BOM fan

beaugalbraith, 2022-03-25:
Hey there,
Is it possible to create a MiniAfterburner toolhead for the Mosquito hotend using the 3007 fan? I printed the Mosquito toolhead. it appears to have been designed for use with a 25mm fan mounted directly to the heatsink. The other toolheads are designed for use with the 3007 fan seated inside the cowling. 

If you look at the toolheads stls you can quickly see that the Mosquito version doesn't have the notch cutout, at the top, that the other toolheads have.

Thanks
-Beau

hyperphonic0, 2022-03-25:
Try this:

https://github.com/hyperphonic0/VoronMods/blob/main/V0.1_Mosquito_Cowling_with_3007_Fan.STL

raymondh2, 2022-03-25:
The toolhead is designed for the 2510 fan intentionally.

closed 2022-03-25 by raymondh2