VoronDesign/Voron-0 #165: End of chain link attachment instructions is missing.

jbeougher7924, 2022-05-04:
Page 90 END LINK? WHAT END LINK?
The end piece of the chain was attached in an earlier step.
If you missed the installation you may be able to mount it to the outside.
A printed cable chain with a small bend radius is included in the released files

However on page 29 shows the end of chain is attached to the bed, however there are not instructions on how to do this.  Page 27 shows the same image but with out the chain link.  The instruction on page 90 are wrong since there is no instruction on attaching this to the bed.  Let alone that brass inserts are needed as well to attach the link to the bed.

ChrisN90, 2022-05-04:
It's not very clear in the manual but it is on page 13 under 'component prep'.

jbeougher7924, 2022-05-04:
Might want to clarify this page better and explain what needs to be done.  Align the page like the other pages.

closed 2022-07-25 by nemgrea