VoronDesign/Voron-0 #166: Manual: some issues I encountered on my build

atlaste, 2022-05-08:
First off, nice work! I really enjoyed assembling my machine using a PIF kit, and am eager to start printing with it. During assembly, I kept note of the issues I encountered, in the good spirit of open source sharing. I only encountered a few issues; hope it helps.

Issues I had with assembly:

1. Not really had an issue with this, but on page 59 it would be nice to say that the notches of the anti backlash nut should match up, so that they the nut stays on tension.
2. Page 64 should make a note about the rotation of the stepper. The kit I had (Fysetc) has the wires cut to the right length, so the connector for the cable should be pointed towards the back of the machine. This is a more general remark btw; I would have liked to know what the rotation was of the A/B stepper as well; during assembly I assumed the cables should go to the back instead of to the sides, which was wrong (and had to fix it later). 
3. Screws that implicitly tap the plastic will destroy the plastic if it's not made to very tight tolerances. I found out the hard way while mounting the pcb brackets (crack...), and manually ran an m3 tap and a 1.5mm bit through the plastic to fix the issue (that's page 128 and 149).
4. Preload 4 nuts for the door handle magnet triangles. Page 38 I suppose, page 150 uses them. It's quite painful to add them later...
5. Belt tension on page 86 is hard to get right without some reference. Is it perhaps possible to use a phone and check the belt frequency?
6. Page 101 with the Z end stop. I just loosely put the nuts on the screws, and then fix them once they are in place. No need to drop them and fight gravity.
7. Page 108 apparently has a different PTFE length than I needed. I needed a total length of 30mm. I would perhaps note that the exact length is very important here; I cut it at 30.2mm at first, and to my surprise it was binding at page 114 because it was 0.2mm too long. It also seemed important to me to cut it with a sharp knife to ensure that the inner size is not smaller than it should be.
8. Getting those fans in place is terrible to get right... Pfew.
9. My thumb screw at page 116 has a separate spring which isn't noted. Feels quite tight. It does appear on the drawing.

That's it. As I said, nothing much for such a detailed design and manual. Hope it helps!

Cheers,
Stefan.

vavaroutsos, 2022-09-15:
I'll add a couple of more issues I had.

1) The lower cable chain end should be mounted to the extrusion using the top hole, not bottom. If you use the bottom hole, the top of the cable chain is a bit too high when the build plate is at Z=0 and it gets caught behind the gantry when moving towards Y=120.

2) The position in the manual for the Z guide rails seems at least 2mm too high. I had about 2mm of rail above the carriages when Z=0 (build plate thumb screws as tight as possible). This didn't allow me to get a full 120mm of Z travel.

3) The left/right position of the Z stage extrusions is critical for some build plates. The PEI spring steel I got with my LDO kit was slightly smaller than 120x120mm. This means any left/right shift in the build plate position (due to extrusion/rail location) takes away from usable build area. I lost a little over 1mm due to my build plate not being perfectly centered. This would not be an issue if the PEI spring steel were oversized 130x130mm.

HitLuca, 2022-12-30:
I took the feedback from @atlaste and updated the manual with some very simple annotations. @vavaroutsos , if you could give me the exact locations of the issues/tips you encountered I will update it with your suggestions as well.
I also added a note on which thread locker to use, as well as integrating the tip on cable routing from [this comment](https://github.com/VoronDesign/Voron-0/issues/176#issuecomment-1325972974)

Keep in mind it's a 5 minutes job and I hope it will be made obsolete by an update to the manual. I also had to compress it since the max file size is 25/26MB and the manual is 27 -.-

[Assembly_Manual_0.1_compressed.pdf](https://github.com/VoronDesign/Voron-0/files/10325070/Assembly_Manual_0.1_compressed.pdf)

vavaroutsos, 2022-12-30:
Referring to my previous comment:

 

1.	Page 36
2.	Page 18
3.	No page, probably an issue with the counter bore depth on the thicker LDO build plate.
4.	Page 25

 

Regards,

 

Pete

nemgrea, 2022-12-30:
we wont be making updates to the v0.1 manual anymore, we are focusing on the v0.2 manual which incorporates everything you have all mentioned here. thank you for the feedback.

HitLuca, 2022-12-30:
Well that's great news! Looking forward to see what the 0.2 improvements are :)

Anacardium12, 2022-12-30:
Is there already a known release date for the V0.2?

raymondh2, 2022-12-30:
There is no announced release date.

closed 2023-01-08 by nemgrea