VoronDesign/Voron-0 #168: SKR Mini E3 V3.0 Recommended Wiring Schematic

mwilson1982, 2022-09-01:
It would be nice to have a recommended wiring diagram like in the manual (which is for the SKR Mini E3 V2) as formbot are supplying this main board in their kit, should be relatively simple to knock up.

nemgrea, 2022-09-06:
we are no longer making updates to the v0.1 manual, recourses are being used for the next v0.2 manual at this time.

closed 2022-09-06 by nemgrea

mwilson1982, 2022-09-06:
> we are no longer making updates to the v0.1 manual, resources are being used for the next v0.2 manual at this time. 
> 
> 
> 
> 

Ok great, thanks for getting back to me, I managed to work it all out, was just trying to save some others of similar issues, may I ask when V0.2 may be getting released?

raymondh2, 2022-09-06:
> may I ask when V0.2 may be getting released?

V0.2 will be released when it's finished. We don't have a timeline or deadline to release it. We are a bunch of hobbyists that work on this project in our free time.

mwilson1982, 2022-09-06:
> > may I ask when V0.2 may be getting released?
> 
> 
> 
> V0.2 will be released when it's finished. We don't have a timeline or deadline to release it. We are a bunch of hobbyists that work on this project in our free time.

Thanks for the reply, sorry just wondering, wasn't pushing for it to be released. If you need any help with any CAD work let me know, I am able and willing to contribute if required.