VoronDesign/Voron-0 #176: Manual Missing Critical Step

suprchunk, 2022-10-31:
Page 96 shows the installation of the cable chain. It shows the correct mounting for the top to the bed mount. It does not show it being attached to the bottom. It should show this. That is problem 1.

Problem 2: it doesn't mention shortening the chain to ensure that the y-stop isn't triggered/broken or whatever will happen when the bed is raised an one is no longer able to zero the machine.

The cable chain is obviously in need of shortening, however one does not realize this until the machine is constructed already.

There is also a need to rewrite the manual, as a lot of information is missing, not touched on, or somehow assumed. Just mentioning this one issue that I ran into that causes a serious issue; unless blocking a stop, or setting it off prematurely is intended.

nemgrea, 2022-11-01:
installation of the other end is shown on page 36. 

the length of your chain depends on the supplier that you buy that item from. we will consider adding a note to the next revision. 

> as a lot of information is missing, not touched on, or somehow assumed

if you have specific things you struggled with that would be helpful...general statements like this dont do us much good though if we dont know whats missing we dont know what to add...

eadz, 2022-11-24:
Personally I found the manual was a bit of an IQ test with puzzles to fill in the missing information in places. 

Two things I did wrong, that were not explicitly mentioned in the manual: 

Orientation of the plug for the A B motors  - the diagrams do not show this anywhere. Turns out the plugs need to face inwards. I had to redo this part. 

The second thing is the wiring of the hot end wires, and how they go down through the gap between the z rails before going under the mentioned side gap. on page 91 there are two highlighted cable paths, but I believe that the cables also need to run through the gap at the back of the head bed? 

[image]

Overall though, a pretty fantastic achievement to make a manual that can build a complex printer with an Ikea-style manual!

closed 2023-01-08 by nemgrea