VoronDesign/Voron-0 #213: Why is there no support for the Mosquito hotend?

green-forg, 2023-01-16:
Just simply out of curiosity, why there is no support for the Mosquito hotend? Is this because of mechanical issues or something else? Sincerely look forward to any replay, and appreciate your guys' contribution.

nemgrea, 2023-01-16:
because i didnt want to make a mount for it. 
I have no idea if it works or has issues as it was never tested with the new toolhead by the dev team. 
the cad is available so anyone is free to make their own mosquito mount.

closed 2023-01-16 by nemgrea

hyperphonic0, 2023-01-16:
@green-forg Here you go, it works great (so far) on the Mini Stealthburner.
Use the longer 2.5mm fan screws through the face with 'Mosquito' written on it for added stability.

[Hotend_Mount_Mosquito.stl.zip](https://github.com/VoronDesign/Voron-0/files/10427937/Hotend_Mount_Mosquito.stl.zip)

green-forg, 2023-01-17:
Wow! That's nice! Thank you for your replay!