VoronDesign/Voron-0 #223: Wrong Link in Manual

TomReinhold, 2023-01-20:
On Page 144, the first part of the linking underneath the QR-Code is different from the written text.
[image]

nemgrea, 2023-01-20:
its showing as correct on my end, what QR code reader are you using?

[image]

millerlarson, 2023-01-20:
I just checked on a fresh copy of the manual and see the same.  Hovering over the QR code itself shows the correct link, but if you hover over the displayed hyperlink itself (below the QR code) the actual link changes to a different one.

nemgrea, 2023-01-20:
ok yes the hyperlink is redirecting incorrectly thank you, we will update it.

the text and the QR code are both the correct link. somewhere in the copy past process the text was changed but the hyperlink was not

nemgrea, 2023-01-22:
fixed https://github.com/VoronDesign/Voron-0/pull/227

closed 2023-01-22 by nemgrea