VoronDesign/Voron-0 #230: File Voron-0/STLs/Raspberry_Pi_3b_Mount_x1.stl not downloadable

Anacardium12, 2023-01-24:
Hi,
The file
[[Voron-0](https://github.com/VoronDesign/Voron-0/tree/Voron0.1)/[STLs](https://github.com/VoronDesign/Voron-0/tree/Voron0.1/STLs)/Raspberry_Pi_3b_Mount_x1.stl](https://github.com/VoronDesign/Voron-0/blob/108d12165d6cbb486b8bc874d380a8e3dd3c9fc1/STLs/Raspberry_Pi_3b_Mount_x1.stl)
Can not be downloaded.
It appears only a lot of text in the browser. All other STL files I tested had no isssue. Tested with multiple browsers.
Pls. check.

---
Text starts like this:
---
solid Raspberry_Pi_3b_Mount_x1
   facet normal -4.061861e-02 -9.991747e-01 -0.000000e+00
      outer loop
         vertex -2.243679e+01 -4.757382e+01 5.000000e+00
         vertex -2.243679e+01 -4.757382e+01 4.000000e-01
         vertex -2.177040e+01 -4.760091e+01 4.000000e-01
      endloop
   endfacet
   facet normal -4.061861e-02 -9.991747e-01 -0.000000e+00
      outer loop
         vertex -2.177040e+01 -4.760091e+01 5.000000e+00
         vertex -2.243679e+01 -4.757382e+01 5.000000e+00
         vertex -2.177040e+01 -4.760091e+01 4.000000e-01
      endloop
   endfacet
   facet normal 1.259666e-01 -9.920345e-01 0.000000e+00
      outer loop
         vertex -2.110877e+01 -4.751689e+01 4.000000e-01
         vertex -2.177040e+01 -4.760091e+01 5.000000e+00
         vertex -2.177040e+01 -4.760091e+01 4.000000e-01
      endloop
   endfacet
   facet normal 8.922681e-02 -7.026948e-01 -7.058744e-01
      outer loop
         vertex -2.119206e+01 -4.712566e+01 0.000000e+00
         vertex -2.110877e+01 -4.751689e+01 4.000000e-01
         vertex -2.177040e+01 -4.760091e+01 4.000000e-01
      endloop
   endfacet

timmit99, 2023-01-24:
That's not an issue. Right click the text file and save as an STL file. It just means that GitHub is thinking that that specific file is a text file and not a binary STL for some reason.

Anacardium12, 2023-01-24:
Worked!
Thx for the quick help.

Anacardium12, 2023-01-24:
Solution found.

closed 2023-01-24 by Anacardium12