VoronDesign/Voron-0 #243: LED Wire routing on MiniSB needs improvement.

MapleLeafMakers, 2023-02-02:
The LED wire routing channels on the back of the MiniSB are very difficult to use without kinking or damaging the wires. 

[image]

In the photo above I've actually already tried moving the slot upward to give more space but it's still insufficient.

Below is the design I was using during the beta, before the official wire routing was added.  It's inspired by the wire routing tabs in the Stealthburner, where a small support is removed allowing wires to be slid underneath from the side.  I've found this to be a significant improvement over the existing ones.

[image]

Step file of the miniSB with this change is attached:

[Mini_Stealthburner_LED_Routing.zip](https://github.com/VoronDesign/Voron-0/files/10572021/Mini_Stealthburner_LED_Routing.zip)

timmit99, 2023-02-02:
I'm using this and it is a nice improvement over the original. Gives more wiggle room for the wires, especially when running 3-4 for neopixels. 😬

nemgrea, 2023-02-08:
Neopixels on the hotend is comfortably in the mods territory. This should be submitted into the user mods repo please.