VoronDesign/Voron-0 #255: Manual, page 49: Wrong text description for microswitch wiring

settinger, 2023-02-20:
Related to #247 , this text box on page 49 of the manual (version 2023-02-10) says "normally open" but shows a *normally closed* configuration. Using the outer contacts like this creates a normally closed configuration on the Omron D2F-5L (the suggested part in the BOM) as well as every other microswitch I've ever seen in my life.

[image]

Thank you to everyone who helped me navigate the Discord and get this issue posted in the right place!

nemgrea, 2023-02-20:
Fixed in source, next pr will include this fix

nemgrea, 2023-03-03:
https://github.com/VoronDesign/Voron-0/pull/259

closed 2023-03-03 by nemgrea