VoronDesign/Voron-0 #260: Missing washers in BOM

martingim, 2023-03-06:
I think there are 8 washers missing from the BOM for the v0.2. It uses 8 washers more than the v0.1 to attach the A and B stepper motors. (page 73 and 79 in the v0.2 manual). But the number of washers in the BOM for the V0.1 and V0.2 are the same.

nemgrea, 2023-03-13:
there are 43 washers needed in the v0.2, the quantities in the BOM have been rounded up to whole numbers to prevent people from buying too few or being stuck if they have a deformed screw/washer. the increase from 35 to 43 was still covered by our listing of 50.

martingim, 2023-03-13:
When I look at the bom on the vorondesign.com website it still says 35 shim washers. Am I missing something?

nemgrea, 2023-03-13:
> When I look at the bom on the vorondesign.com website it still says 35 shim washers. Am I missing something?

this should be fixed now, sorry for my misunderstanding and thank you for the report!