VoronDesign/Voron-0 #271: Sourcing Guide M2 SHCS "Alternative source" URL points at a BHCS listing

ditdafivo, 2023-04-13:
In https://vorondesign.com/sourcing_guide?model=V0.2  the rows for both M2x6 SHCS and M2x8 SHCS (ISO-7380-1) list the following link for the "Alternate Source" column:  https://www.google.com/url?q=https://www.aliexpress.com/item/32966917947.html&sa=D&source=editors&ust=1681365417744949&usg=AOvVaw2Vlay7R7Q7xqN1y8BuRLXa 

Opening that link, though "Socket Head" is among the words in the title, the picture of what is being sold are clearly BHCS. I overlooked that, ordered them and indeed was sent BHCS.

ditdafivo, 2023-04-13:
The ISO reference on those lines should also be ISO 4762 / DIN 912 , not ISO-7380-1 which is for BHCS. 

This compelled me to confirm that pages 21,25,41,113,134 of the manual do all want M2 SHCS for the carriage/rail mounting, not M2 BHCS.

ditdafivo, 2023-04-13:
I'm considering the following from aliexpress for DIN 912 

Black Steel:
https://www.aliexpress.us/item/3256802174163152.html?spm=a2g0o.productlist.main.17.749765e6B6F8mG&algo_pvid=04eaaeaa-fcec-4419-862c-57a0357dc85a&algo_exp_id=04eaaeaa-fcec-4419-862c-57a0357dc85a-8&pdp_npi=3%40dis%21USD%214.96%211.49%21%21%21%21%21%400b0a558a16813637057553191d0794%2112000020310654879%21sea%21US%211812182282&curPageLogUid=eZJmGonwpNxy

Or Stainless, though M2x6 is currently out of stock:
https://www.aliexpress.us/item/2251800572951561.html?spm=a2g0o.productlist.main.1.749765e6B6F8mG&algo_pvid=04eaaeaa-fcec-4419-862c-57a0357dc85a&algo_exp_id=04eaaeaa-fcec-4419-862c-57a0357dc85a-0&pdp_npi=3%40dis%21USD%211.39%211.39%21%21%21%21%21%400b0a558a16813637057553191d0794%2112000024596927812%21sea%21US%211812182282&curPageLogUid=dgdM8RgfwAAD

nemgrea, 2023-04-14:
sourcing guide is updated, thanks.

closed 2023-04-14 by nemgrea