VoronDesign/Voron-0 #279: V0.2 manual: High contrast version?

slashbeast, 2023-04-20:
I have significant issue with following the manual's PDF attached in this repository due to problems with contrast. Common tools like zathura allows me to switch colors to negative but this is still not quite there. I know that Voron 2.4 has high contrast manual but I also found those not as accessible as I'd prefer.

I was looking for sources for how those PDFs are generated but found none. Ideally we could have a version that has all the grey (regular) text replaced with black font, slight thicker, red with dark-red and perhaps images with slightly thicker lines, it would not look as appealing as it looks right now, but could help some visually impaired people.

nemgrea, 2023-04-21:
you can open the PDF in any common PDF editor (even online ones) and make any formatting changes you wish

closed 2023-04-21 by nemgrea

slashbeast, 2023-04-21:
This is not very helpful if you expect all visually impaired users to edit PDF and do all of this when those could be readily available to people much like Voron 2.4 has in https://github.com/VoronDesign/Voron-2/tree/Voron2.4/Manual where you have highcontrast variant of manual. 

This will also not allow one to address the part regarding difficult to see inline images, at most you can modify the properties of fonts there in those PDF editors.

slashbeast, 2023-05-13:
Should anyone found it in future, one can bump contrast of the manual using ImageMagic, the downside is that it will first convert every page to jpeg, bump contrast and then save it again as PDF, with DPI of 300 the manual at the moment of writing it goes from 48 to 611 MB however it does provides a significant comfort for those that needs it without adding work to Voron team.

    convert -density 300 VORON_V0.2_Assembly_Manual.pdf -contrast-stretch 0.5% high_contrast.pdf
    
 The process take significant amount of time due to being bound to single CPU thread.