VoronDesign/Voron-0 #280: CAD model missing 6 nuts on the X Gantry Extrusion

SrgntBallistic, 2023-04-21:
Was just looking through the CAD Model. I noticed when I hid the X Gantry extrusion to look at the preloaded nut locations that there weren't any nuts present to screw the X/Y Joints to the extrusion.

The instructions clearly shows pre loading the nuts but thought it could cause some confusion for anyone using the model as a reference for nut locations. Every other proloaded nut seems to be present
[image]

kyleisah, 2023-04-29:
Oops! I will take a look at this. Thanks @SrgntBallistic!

kyleisah, 2023-05-10:
Just want to provide an update. I've added the missing parts to the CAD. Just waiting on some other things to be finalized, and the newest CAD will be pushed. Thanks again!

SrgntBallistic, 2023-06-06:
@kyleisah I think I might have noticed a few more missing nuts.

The 4 for the door latch on the inside of the left front upright
[image]

SrgntBallistic, 2023-06-06:
I have only been able to note these things because I've created a barebones version of the Fusion file with just the extrusions and hardware to help myself (and maybe others) identify were all of the preloaded nuts are.

[image]

nemgrea, 2023-06-07:
https://github.com/VoronDesign/Voron-0/pull/291

closed 2023-06-07 by nemgrea