VoronDesign/Voron-0 #283: Manual page error

ThomasDennisSp, 2023-04-30:
in the v0 manual on page 109 and 110 are the x y joint left side but page 111 right side is the left joint alias page 109 again but on page 112 it is the right joint

nemgrea, 2023-04-30:
disregard the page numbers in my images as im in the process of updating some other items for our r2 release but im not sure im seeing the disconnect here? i believe the xy joints are pictured correctly?

[image]
[image]
[image]
[image]
[image]

ThomasDennisSp, 2023-04-30:
the first and 3rd image you saend are the same both are the left joint exept that the right side is rotaded and not mirrored

ThomasDennisSp, 2023-04-30:
[image]
[image]
[image]

the red circled ones are the same but the 2end one should be mirrored to be the one for the right side
the blue circled ones are the same exept that the 2end one is rotated by 180 degree and not mirrored how it should be if it is the right side one
and then on the 3rd circled in green how it should be if  it is the right joint

nemgrea, 2023-04-30:
[image]

ThomasDennisSp, 2023-04-30:
oh ok then i saw that wrong but i think if the right side is shown from the front it whould be better

closed 2023-05-01 by nemgrea