VoronDesign/Voron-0 #284: Electronics STLs: VHB DIN mount and PCB DIN clip are incompatible

settinger, 2023-05-08:
In the current version of the repo, there is an STL file [`VHB_Din_Mount_x2.stl`](https://github.com/VoronDesign/Voron-0/blob/Voron0.2/STLs/Electronics/VHB_Din_Mount_x2.stl), which replaced an older STL file [`VHB_Din_Mount.stl`](https://github.com/VoronDesign/Voron-0/blob/d9e205dd984bb772386feaa3e9d27dadbdf73b85/STLs/Electronics/VHB_Din_Mount.stl), no longer in the repo.

The issue is that the current file [`PCB_DIN_Clip_x2.stl`](https://github.com/VoronDesign/Voron-0/blob/Voron0.2/STLs/Electronics/PCB_DIN_Clip_x2.stl) does NOT fit the current `VHB_Din_Mount_x2.stl`, but it DOES fit the older, deleted `VHB_Din_Mount.stl`. The angled cleat portion of `VHB_Din_Mount_x2.stl` is too thick, and the portion of `PCB_DIN_Clip_x2.stl` that is supposed to "snap" into place is unable to catch on anything.

If anyone else prints all three STLs and gives them a test-fit to confirm or contradict what I found, I'd greatly appreciate it! I don't *think* this is an issue with my printer tuning, but printers always have a way of surprising me, so who knows? :)

kyleisah, 2023-05-10:
Hey there! The DIN cleats were my idea. I designed them and tested them myself, and they've been released for a few months now. 

But, to see if maybe something was wrong, I printed them again straight from the repo and they worked just fine. 

Here's some photos, maybe you've got something upside down?

[image]

settinger, 2023-05-10:
Thanks for looking into it, and thank for designing the cleats in the first place! That's definitely the same cleat orientation that I was using. I've attached a video of clip and the cleat that I printed yesterday, and how they pop apart with very little force. The original cleat, by comparison, is very secure; I'd film a demo of that part for you, but it's already sealed up in my V0 build, haha!

https://github.com/VoronDesign/Voron-0/assets/7494643/3a74da9c-3dba-4304-8ef0-3f05f025b772

Here's a cross-section of the old cleat (left) and current cleat (right). Perhaps they are just different enough that it only makes a difference on a less-well-calibrated printer like mine? If that's the case, I can see the argument for not needing to alter the design, but it's strange because the older version worked so, so well for me!

[image]

kyleisah, 2023-05-10:
I'm curious if what you're having is due to thin wall settings or something like that. It looks like your DIN clip is sort of rounded where there should be a small protrusion from the front leg. 

Do you have a sliced gcode file of the PCB DIN clips and the current rev of the cleats?

If that something like that is due to a slicer issue, I definitely want to help improve it!

And thank you! KAMP is a lot of fun, and super useful. 😄

settinger, 2023-05-10:
Sure thing! [Here are the sliced gcode files for the clip and both cleats.](https://github.com/VoronDesign/Voron-0/files/11438028/VHB.DIN.gcodes.zip) These ones are printed in PLA, but the ABS ones I printed a few weeks ago have the same fitment. Which clip leg is the front leg you mentioned? I'll get some still images in the morning if that'll help.

kyleisah, 2023-05-20:
@settinger Hey, just getting back to this. I took a look at the gcode files you shared and compared them to mine, and they look the same. The pieces that I printed, I agree, could feel a bit better. They don't feel as strong as I think they should. I'll see if I can dig up some old files or just redesign a new clip. If you'd like, I can send an STL over to you for you to try?

kyleisah, 2023-05-20:
[image]
I think something like this could be kinda fun 😄

kyleisah, 2023-05-20:
[image]

These seem to clip a bit better and hold a bit better. But let me know please, so we can get them into the repo. 😄 

[VHB_Din_Mount_Fix.stl.zip](https://github.com/VoronDesign/Voron-0/files/11521239/VHB_Din_Mount_Fix.stl.zip)

settinger, 2023-05-20:
Rad! I'm loading up my ABS now!

settinger, 2023-05-20:
Now _that_ is a secure clip! I printed all three VHB DIN clips in one go: the old one that worked for me previously, the one that's current in the repo, and your new one with the cool logo. Sure enough, the oldest one and the newest one lock in place perfectly. Thanks a ton for whipping that up, I hope some other folks can print these and chime in with their results as well!

closed 2023-06-07 by nemgrea

dtdionne, 2023-09-17:
Hmmm, I have a similar issue.  It's like the backstop is almost a mm too thick or tall.  The cleat doesn't snap in.  I did the best I could trying to interpret voron print settings into orca slicer.  I haven't had this kinda trouble with any other parts and I printed all my own parts for the 0.2.

But i do think there's a "thin wall" check box.  I've already printed a dozen of these in different materials to test...what's a few more.

dtdionne, 2023-09-17:
I just watched that video and the ones I just printed don't even fit that good.  I shaved a little over .5mm off one and it snaps in now.  But mine cleat also seems more rounded where it should (and probably is) sharper.  I bet it's a print setting in orca.