VoronDesign/Voron-0 #286: Request to add printed parts list at begining of each section.

Faenum, 2023-05-26:
Im printing the parts as I go, and the manual doesnt always tell you what the part is called. Im having to find the part on the cad to find what its called, before I can find them in the files to print. Adding the names of the parts onto the render at the begining of each section would help make things faster and more straighforward.
Ive included a screenshot of the page Im reffering to.
Thanks
[image]

Faenum, 2023-05-26:
Ive also just discovered that the names dont always match up between the CAD and the files

kyleisah, 2023-06-26:
A lot of that is much easier said than done. The CAD has since been updated with named bodies that match the STLs in the repo. (This also saves us some hassle on export of STLs, so it's a win-win.)

I will not be adding the names of the section's parts to the renders as the renders follow a design aesthetic. The render already shows you what your printer will look like at the end of the section, so you can just look at the machine in front of you, and look at the render, and know what is being added in that section. 

Lastly, having a list of all BOM items per section is kind of a big ask. All of that information would have to be exported from the CAD, so you would have to manually select and pick through what all is being added in that section, and feed that to a BOM extraction script and hope you don't miss something or you'll be doing it all again. 

I understand those things would be nice, but ultimately, they would clutter up the manual when all of this information is *technically* already provided for you. 

Keep in mind, Voron is not a company. We're just a bunch of hobby nerds making printers we like, and giving it away for free.

Faenum, 2023-06-27:
Thanks for the update on the CAD, that will make it easier to find what Im looking for.

It doesnt need to be on that page, that was just a sugestion. I dont mean links to the parts, I just mean a text label next to the parts with the name of them, maybe the same as the fixings are labled. Because as it is now, its long winded to have to load up the CAD and search around it to find the part name, then try and find it in the STLs list to print it, when it would be much easier to have the name of the part on the manual in the section where its used.

kyleisah, 2023-07-07:
I don't think that it's necessary. We follow a design scheme for these manuals. You don't have to load up the cad, just dig for the part and you're good to go.

Faenum, 2023-07-07:
Thats the issue though, I cant dig for a part ive not printed yet, Ive been printing parts as I need them, Im not able to work on building it every day so use those days to print the next parts. Its difficult to find the part I need in the parts folder when I dont know what theyre called.

nemgrea, 2023-07-07:
are you having trouble opening the printed parts guide? that tells you exactly what each part is by name and on what pages they appear. theres also a handy picture for every stl so you know what they look like?

Faenum, 2023-07-08:
I didnt know that existed. Thanks