VoronDesign/Voron-0 #290: V0.2 manual: Clarify the preload of nuts on page 27/29

slashbeast, 2023-06-05:
The page 27 mention preloading two nuts and confirms it on page 29, however user might misunderstood it and think that the nuts are for the m3x6 BHCS screws and therefore go forward until the final nuts check is to be done and realize that those nuts are not there.

It is mentioned that all screws goes into nuts unless told otherwise, but I think a simple note of something along the lines "Preload 2 nuts for the total of 4 nuts in this side of extrusion' could help. I fall into it and were not fan of taking apart the entire frame to preload those nuts, thankfully I had the m3 slide-in nuts on hand.

nemgrea, 2023-06-06:
[image]
this has been fixed already, will be present in the next revision

slashbeast, 2023-06-06:
Perfect, many thanks.

closed 2023-06-07 by nemgrea