VoronDesign/Voron-0 #296: Voron0.2r1 - [a]_MiniSB_Cowling_x1 HE Fan face only 2 layers (0.4)

nigelpjames, 2023-06-12:
The front of the cowling around the HE fan meant to only be 0.4 thick?  Seems rather thin and wonder if this is a mistake.  In teh previous release, the front had 4 layers (0.8)?

New 0.2r1 on left, 0.2 on the right, layer 3

[image]

nemgrea, 2023-06-12:
the current STL's inm the repo are 3 layers thick. looks like you downloaded the files before the PR was fully merged, this has been fixed

closed 2023-06-12 by nemgrea

nigelpjames, 2023-06-12:
no, i only downloaded the repo 12hrs ago from the default r1 branch.  the files are showing as PR #291 which doesn't mention  the fix and both cowls still appear to only have 2 layers

nemgrea, 2023-06-12:
Check your slicer settings the part thickness is 0.6

nigelpjames, 2023-06-12:
Thanks, all good now...slightly thicker first layer so much have been on the margin since its still thinner than the 0.8 in RC2

nemgrea, 2023-06-13:
V0.2 Master Assembly RC2.zip
this is not the cad file that appears in the github....

scul86, 2023-06-13:
Copied the wrong filename.

Reverified that I was using this file:
[V0.2R1 Master Assembly v62.step.zip](https://github.com/VoronDesign/Voron-0/blob/Voron0.2r1/CAD/V0.2R1%20Master%20Assembly%20v62.step.zip)