VoronDesign/Voron-0 #302: Change assembly manual - let Z motor be assembled earlier

Matszwe02, 2023-06-20:
Move z motor assembly from page 107/108 to page 58. Mounting the Z-motor with the bottom acrylic panel is a pain - a single allen key barely fits, and if someone has some electric or other fancy screwdriver it can't fit at all. Aligning the nuts and placing bolts in place is very hard when the acrylic is almost directly above them. Let users mount Z-motor before the bottom panel is slided in place.

Kagee, 2023-11-27:
Just mounting the Z motor will make the printer quite unwieldy. Could one maybe move the whole z-motor + feet to before the gantry part?

TheRealDeathsneeze, 2024-01-17:
Seconded. installing the Z assembly and bed frame (but not the bed just yet) is substantially easier to do before the deck and mid panels have been installed and can be done prior to assembly the XY motion system.