VoronDesign/Voron-0 #304: Config - Sensorless Homing Current & Macro Confusion

SrgntBallistic, 2023-06-24:
I had some issues setting up Sensorless Homing for my Formbot V0.2(r0) Kit.

I used the [`bigtreetech-skr-pico-v1.0.cfg`](https://github.com/VoronDesign/Voron-0/blob/Voron0.2/Firmware/bigtreetech-skr-pico-v1.0.cfg)

While also walking through the Voron Docs Initial Startup guide and Klipper TMC Docs.

I first walked through all the instructions in the .cfg file.

I set the motor currents based on the recommended equation in the file. 

["# you need to calculate the run_current value using the equation (rated_motor_current * 0.707 = Maximum_run_current) start with a value that is about 60%-70% of your maximum value."](https://github.com/VoronDesign/Voron-0/blob/a4d02db92a7dd71e8f2c72d6f74f6bc3b9d020c8/Firmware/bigtreetech-skr-pico-v1.0.cfg#L74)

My Moons motor currents are:
- A/B - 1.5A
- Z - 0.65A
- E - 1A

Using the above equation and factoring a 65% ( x0.65) starting value, I ended up with the following currents
- X/Y - .69
- Z - .3
- E - .46

When I got the the `_HOME_X/Y` macros I first found the wording a bit strange for changing the `XX` value. It states:

["{% set HOME_CURRENT = XX %} # replace the XX with a current value that is about 70% less than your run_current value for X and Y "](https://github.com/VoronDesign/Voron-0/blob/a4d02db92a7dd71e8f2c72d6f74f6bc3b9d020c8/Firmware/bigtreetech-skr-pico-v1.0.cfg#L406C5-L406C134)

By 70% less does this mean to set XX to 30% of our calculated `run_current`? That's what I took it as.
.69 * 3 = .21 which is a very low current setting.

After doing so I had a lot of trouble getting a stable Stall Sensitivity as outlined in the Starting Guide.

Taking a look at the macro closer I noticed it's actually taking that XX value and setting it as the `HOME_CURRENT` value. But then sets the current to `HOME_CURRENT * RUN_CURRENT` for homing

So it seems to me that it is actually looking for a decimal percentage?

Looking at the macros presented in the [Voron Sensorless Guide](https://docs.vorondesign.com/community/howto/clee/sensorless_xy_homing.html) and the [Klipper TMC Sensorless Homing Macros](https://www.klipper3d.org/TMC_Drivers.html#using-macros-when-homing) section I noticed that neither do any multiplication. They just directly set the current to a hardcoded value for homing and change it back.

So it's possible I've just really confused myself here and read things wrong. Or is the intention to change the `XX` value to a percentage that will be multiplied by the `RUN_CURRENT` to drop the current during homing? Or lastly is the goal to have users manually set a current value that's lower than the set `RUN_CURRENT`?

kyleisah, 2023-06-26:
Hey there, just letting you know I will be taking a look into this and see if there's something better I can come up with that works for newbies and pros.

nemgrea, 2023-06-26:
https://github.com/VoronDesign/Voron-0/pull/305

closed 2023-06-26 by nemgrea

SrgntBallistic, 2023-06-26:
> Hey there, just letting you know I will be taking a look into this and see if there's something better I can come up with that works for newbies and pros.

Thanks. I definitely saw a couple other folks in the discord channel having questions about that.

One of those things were ppl are likely figuring out what it means and how it works. But the clarity can just short circuit some of the process to get to a working value.

Thanks for the work! The config worked great to get started!