VoronDesign/Voron-0 #306: Add clearance between hotend and x carriage

Matszwe02, 2023-06-29:
[image]

I have dragon hotend and my mini stealthburner is tilted a little because the hotend presses against the ends of the belts. I needed to trim these belts to be flush with x carriage and only then there was no issues with the stealthburner.

nemgrea, 2023-06-29:
this issue is addressed on page 142 of the assembly manual 
`trim the belts 
to their final lengths, the amount you can leave depends on the hotend you are using. If your hotend sits flush against the X carriages front facing surface a flush cut may be required.`

there is simple no extra room to add space for oversized heatsinks like those used on the dragon toolhead.

Matszwe02, 2023-06-29:
> there is simply no extra room to add space for oversized heatsinks like those used on the dragon toolhead.

And I think that's the issue: someone explained in voron's discord that textured pei can add the minimal amount to z height, which is sufficient to cause issues with the clearance. Maybe extend the z carriage or anything else by let's say 0.2 or 0.4mm so it will fit better?

nemgrea, 2023-06-29:
the toolhead will hit the front door if the x carriage is extended.

Matszwe02, 2023-06-29:
> the toolhead will hit the front door if the x carriage is extended.

ok, right. Is it possible to make some belt guides to fold away or there's just no way it can fit?