VoronDesign/Voron-0 #308: Guidler has too much play - possible solution

Matszwe02, 2023-07-04:
Ministealthburner has an issue that the amount of play in the guidler is significant enough to cause issues when extruding. It could grab filament too weak for fast printing, and everything above 5mm³/s wasn't possible.

https://github.com/VoronDesign/Voron-0/assets/49353902/af5799f3-7e6b-418b-82b3-9f7547d47321

I made some changes on the guidler that I think will fix it - check it out.
I narrowed the bold hole from 3.2mm to 3.1mm (which is still enough for even less accurate printers) and applied a tiny angle to push it towards filament. In that way the remaining wobble that may still be there will be directed towards the filament, not away.

[V0.2 Guidler v1.f3d.zip](https://github.com/VoronDesign/Voron-0/files/11952082/V0.2.Guidler.v1.f3d.zip)

nemgrea, 2023-07-05:
Are you sure youre extruding at 5mm cubed per second and not just using the preset numbers in mainsail that are linear feed rates and not volumetric flow rates?
[image]

Matszwe02, 2023-07-05:
> Are you sure youre extruding at 5mm cubed per second and not just using the preset numbers in mainsail that are linear feed rates and not volumetric flow rates? [image]

yes, I was looking at this parameter:
[image]

After this mod i'm no longer extruding max 5mm³/s, but I'm now maxing out the hotend at 15mm³/s. Further speeding up actually forces extruder motor to skip instead of gears grinding. I may later install cht nozzle, which will push these limits even more, but now I'm sure that the guidler doesn't cause any issues.

probably-Erwins-Cat, 2024-01-26:
Shameless commenting here that your issue might be mine in disguise: https://github.com/VoronDesign/Voron-0/issues/340.

Matszwe02, 2024-01-27:
> Shameless commenting here that your issue might be mine in disguise: #340.

My post is about bad tolerances that lead to too much play on the guidler (I never experienced thermal deformation, just the high amount of play that this part has), your post is about the temperature that is badly managed - so I think 1 part, 2 issues

nemgrea, 2024-04-30:
closing do to updated issue being submitted

closed 2024-04-30 by nemgrea