VoronDesign/Voron-0 #315: Distance between Lower Stain Relief Body Screws and Top A/B Mounts too small to account for compounding tolerances

ChaosBlades, 2023-08-08:
I noticed that the screw head is ever so slightly colliding with my top A/B motor mounts. I would hazard a guess this is the case for several people and they just have not noticed. I ended up taking a file to the top A/B mounts to make clearance. Compounding tolerances is the cause of this. Such as over sized M3 screw heads, linear rail heights, extrusion dimensions, and the print head slighting rocking back when it gets tightened due to again compounding tolerances in printed parts. This could cause inconsistent sensor less homing issues.

Lower Stain Relief Body screws should be raised by 1 mm to avoid this issue or could be raised by 0.5 mm and the top of the A/B motor mounts lowered by 0.5 mm.
[image]

nemgrea, 2023-08-08:
ISO 7380-1 tolerance for m3 screw heads is a max value of 5.7mm so the cad is already at the max material condition for the screw. 

dimensional tolerances for MGN7 rail heights are at most +/-0.04mm (class C precision)

ChaosBlades, 2023-08-08:
My original V0 kit came screw heads of up to 5.95 mm I didn't use them for this though. My original rails were actually dead accurate. Can't say the same for the bearings though lol. I have since switched to KB 3D. Have not measured those. Printed parts and maybe angle and positioning of heat insert are probably the biggest contributing factors. Regardless of the exact cause simple enough issue to resolve in the next version.