VoronDesign/Voron-0 #336: Do the Raspberry PI 3b and pico need separate power for usb connection?

tianyi725, 2023-12-16:
[image]
Connect the Raspberry PI and pico motherboards using usb.

I read in the pico manual that using a usb connection requires powering the Raspberry PI separately, is that right? (Red circle on the picture)

I see the voron0.2 pdf does not say that this connection also needs to be powered separately.

I also don't see how the Raspberry PI is powered on the voron0.2 pdf?

I have an idea,
Or use usb to connect the pico and Raspberry PI, and connect the uart cable to power the Raspberry PI, is that OK?

nemgrea, 2023-12-17:
Power to the Rpi is shown on page 209.

[image]

closed 2023-12-17 by nemgrea

tianyi725, 2023-12-17:
> Power to the Rpi is shown on page 209.
> 

I contacted pico tech support.

Tell him to use a usb connection and uart power, okay?

He told me: It works in theory, but it may not be stable