VoronDesign/Voron-0 #339: Page 104: Unclear text about the filament sensor

gregersn, 2024-01-20:
The text in the manual, on page 104, states: "Prepare the switch for the Filament runout sensor by soldering wire to t**he two outer terminal**. " (my emphasis)

However, the picture of the switch has the two rightmost terminals indicated by a darker colour.

From what I can see on my switch, the colouring is the correct way to wire the switch normally open, but the text confuses this point, unless you by "outer" mean the terminals closest to the outside of the printer, but that is not really all that easy to deduce from the picture.

nemgrea, 2024-01-21:
good catch itll be updated in the next version of the manual thanks!

nemgrea, 2024-06-09:
https://github.com/VoronDesign/Voron-0/pull/354

closed 2024-06-09 by nemgrea