VoronDesign/Voron-0 #340: Guidler prone to slight deformation under higher temperatures, resulting in loss in gripping power

probably-Erwins-Cat, 2024-01-26:
hey guys,
the current guidler is not sufficiently reinforced to keep straight under load and higher temperatures. After deformation, it shows the same failure mode as https://github.com/VoronDesign/Voron-0/issues/308 that you can pull the filament out of the extruder without the groves of the gear properly gripping the filament, i.e., the filament slips through the gear teeth without turning the extruder gears and motor.

I have reproduced my issue a few times, two times involuntarily, one time with my full intention. A rundown:
While printing in a hot environment, i.e., high chamber temp of approx 55°C+, with a heated bed, hotend, and motor, the extruder gear parts, including the guidler, are getting quite warm, warmer than the chamber temp. This can reach the heat deflection temperature (HDT) of some softer ABS/ASA filaments, resulting in a slow creep of the non-reinforced side of the guidler, see images.

[image]

[image]

This causes the gear to misalign. Its rotation axis is no longer parallel to the other gear rotation axis. Higher tensioning cannot resolve this issue (I tried it) as the gear side is fully engaged, but the gripping grove side is moved farther apart. When gently pressing the deformed corner, the extruder immediately works again (as long as you keep pressing)

Similar to the issue https://github.com/VoronDesign/Voron-0/issues/308 I thought it was the too large hole diameter, but it is not. Yes, the reduced play might help a little, but the issue comes back when upper high-temperature and low HDT conditions are met.

Things I tried out and solved the issue: 
- take every precaution to reduce temperature towards the guidler, e.g. lowest reasonable chamber temp, hotend and motor current. This, however, limits the printer's capabilities (indirectly). The z-layer adhesion and bed adhesion would suffer.
- switching over to material of higher heat deflection temperature. Unfortunately, many are limited to printing ABS/ASA and many community go-to ABS(+) have shown QC/QA issues and lower HDTs. PC-ABS has high enough HDT but can't really be printed on average-joe printers. My PC-ABS guilder still holds
- modifying the guilder to accept (non-self-tapping) M2x16 as reinforcers. This helps to keep the guilder straighter, i.e. the M2x10 self-tapping screws are too short and do not dig deep enough into the "meat" of the guidler

Things that weren't sufficient:
- lowering the tension to a minimum. The force at the lowest possible tensioning that still pushes the filament reasonably is still too high when the HDT is reached.

Recommendations and things to look into:
- Take a look at the guilder again and check if changing the M2x10 self-tapping screw (that has an effective length of 9mm) can officially be changed to something like M2x16.
- Somehow, make the lower part beefier to resist deflection. This requires a change of other parts to make room for that.
- intentionally add a small tilt of the rotational axis of the gear of the guidler for compensation. Alternatively, the side where the tensioning knob and the latch meet can be slightly wedged, so that the tensioning knob also induces a slight torsion to the guidler for compensation.

fhupfi, 2024-03-01:
Experienced the same issue.

As it is right now, the pink marked portion of the guidler is to weak for the usage in enclosed chambers, when printed from ABS.
- A M2 Bolt is in my eyes not suitable to take the torsional force. Also it doesnt solve the problems root cause.
- i fully support your solution to make the guidler beefier.
- most important: take the torsion (green Arrow) out of the guidler, by bringing the orange marked Forces (pivot point, hobb of the BMG Gear tensioning screw) as close as possible in the dashed red line.

[image]

NerdyGriffin, 2024-06-19:
This issue is made worse by the fact that the anti-squish screw is even farther from the tensioner spring than the filament path is from the tensioner, so the torsion @fhupfi describes gets worse when using flexible/squishy filament. I attempt to describe this in a comment in another issue https://github.com/VoronDesign/Voron-0/issues/348#issuecomment-2178829877

nemgrea, 2024-12-18:
duplicate of #348. closing this one as more discussion has occured on the other entry

closed 2024-12-18 by nemgrea