VoronDesign/Voron-0 #347: Need easier way of accessing extruder grubscrew

ccfman2004, 2024-04-16:
There really needs to be way to make accessing the extruder grubscrew in the Mini Stealthburner much easier.

I cannot at all see the grubscrew when the manual tells me to loosen the grubscrews to get the gear properly aligned with the filament path. All I can do is blindly stick my hex driver in while turning the 50T gear and feel around which is not going well at all as the hex driver is very difficult for my large fingers to grip without dropping it. Because I can't see what I'm doing I don't even know where exactly the grubscrew is in relation to the 50T gear. There isn't even enough space for me to even get a light in there as the Guidler part creates a shadow. I've spent the last 3 days trying to loosen one of the grubscrews with no success.

nemgrea, 2024-04-16:
[image]
[image]
[image]

ccfman2004, 2024-04-17:
I don't understand how you can't get so much light to make it so easy. I can't seem to do that. LOL.