VoronDesign/Voron-0 #348: Guidler upgrade proposal

Matszwe02, 2024-04-19:
I posted an [issue](https://github.com/VoronDesign/Voron-0/issues/308#issue-1788425942) previously, and now I upgraded the suggested idler design even more

- the standard one couldn't hold very flexible TPU even a tiny bit, now it can grab it so I can't get it out with huge force.

My suggestion implements lowering the axis screw's tolerance and widening the gap on the spring tightening section, thus making the minimal gap between gears low enough to squish even very flexible filaments

Also, this design is based on that previous issue, so also the axis screw's hole is at a tiny angle to ensure that the gear won't twist away from the filament, but rather twist that much so it will be parallel to the other gear

[V0.2 Guidler v4.step.zip](https://github.com/VoronDesign/Voron-0/files/15038331/V0.2.Guidler.v4.step.zip)

Matszwe02, 2024-05-08:
# I will add some graphics to show what this change does exactly.

## shape is very similar to the original one - to be honest it can't be distinguishable. It's changed by fractions of mm in the axis and mating surface.

[image]
[image]

# Front view:
To be honest it doesn't look like a significant change from this perspective

## Before:
> [image]

## After:
> [image]

# Top view:

Here it's visible: 0.2mm vs 0.6mm gap (on the scrren it's 0.2-something, but I didn't count the tiny bit of the axis skew)

- Thus from original 1.4mm filament path, it's now 1mm!
- Counting different print quality from user to user, I can say that this 1.4mm gap can be wider on some printers, being very close to the 1.75mm. I can imagine some tpus easily squish by that 0.3mm.

## Before:
> [image]

## After:
> [image]

nemgrea, 2024-05-09:
the images and measurements in the "BEFORE" screenshot are not accurate. the cad model is modeled in a symetric constrained state that does not represent the actual position of the gears when there is no filament in the toolhaed.

there is a gap between the guidler and the shuttle on purpose. the thumb screw spring is always trying to close this gap. if we add a hinge mate to this component you can see how the guidler will actually be sitting when the spring is tensioned.

[image]

you can also see that this state puts the drive gear and idler gear very close to the maximum possible meshing that the gears can do. (in fact most people can feel the "gritty" feel of those gears bottoming out when they spin the 50T without filament present in the assembly)

we can see below that the available space for the filament in the CAD modeled position is 1.6mm

[image]

however again this is not the minimum position that the gears can reach.

below we can see the same measurement but when the gears are closer to their minimum gap position and we observe that it is much closer to 1.4mm

[image]

lastly to decrease this 1.4mm dimension further no changes to the guidler are needed, we can simply move the face of the shuttle component back and increase the available travel of the guidler swing arm. however remember that there is a hard limit on the minimum gap that the two BMG gears can achieve due to the gear teeth bottoming out and preventing the two parts from getting any closer to each other.

[image]

Matszwe02, 2024-05-09:
Thanks for pointing the inaccuracy in my measurements - I used the fusion default model without rotating the guidler to its desired position. However, that doesn't change the fact that my voron couldn't print reliably with the standard design. One of my proposed changes - tilting the guidler axis, helps with the issue of teeth being too close to each other. Maybe that's only my issue, but I experienced a tiny bit of play in the guidler axis, which caused the little backlash that was enough to lose the grip. So maybe my proposed changes will enhance quality on lower quality guidler prints, as it allows for more inaccuracy than the original one.

If my suggested new guidler will be used, there will be close to no difference with the good-quality guidler, as it's affecting the normal working very little, all it does is to allow for more space to squish softer or thinner filaments.

NerdyGriffin, 2024-06-19:
I also support this change because I had similar issues with the mini-SB and I ended up fixing it by adding a shim washer to the mating surface that @Matszwe02 is suggesting to change. 

I know there are some issues with the way he presented the measurements, but the real-world results of the suggested change do effectively solve a common issue with the current guilder design.
Due to the fact that the "minimum gap" adjustment screw (referred to as the "ANTI SQUISH THINGYMAJIG" in the manual) is over by the filament path rather than inline with the shuttle and spring, there is a long-term issue where the spring-side gap will creep tighter due to the constant spring pressure and eventually cause the extruders gears to mesh too tightly and bind up. If you adjust the anti-squish screw to increase the gap, it will print fine for a while, but over time the spring pressure will warp the guilder and force the gears closer again. This uneven creep/warping can eventually get so severe that you can have the gear-side gap of the drive gears binding up while the filament-side gap is large enough to slide filament straight through as if it were unlatched (regardless of the anti-squish screw position). 

This issue is more common when printing flexible filament because the squishy filament allows all the clamping force to be on the anti-squish screw rather than on the filament itself. The distance between the spring and the anti-squish screw is larger than the distance between the spring and filament path, so the unwanted torque exerted on the guilder is higher (same forces but exerted farther apart = stronger torque around the z-axis to warp the guilder)

**TLDR**:The real issue is not the size of the filament or gear gap as seen in the CAD. Rather, the issue is the fact that this gap changes over time due to the uneven pressure from the tensioner spring and the anti-squish screw.

This is not an issue on the full size StealthBurner because the anti-squish screw in inline with the guilder shutter and spring, so the spring does not exert any unintended twisting force on the guilder. 

What the suggested change from @Matszwe02 effectively does is adjust the physical limit of how closely the gear side of the guilder can mesh when the latch is latched, so that the gears cannot get too close and bind up. In other words, it is compensating for the lack of an anti-squish screw on the gear-tooth side of the guilder.

Ideally, the solution would be to move the anti-squish screw forward so that it is directly underneath the shuttle (like in the full size StealthBurner), but that may be difficult or impractical in miniSB due to the size constraints.

Matszwe02, 2024-06-25:
> I also support this change because I had similar issues with the mini-SB and I ended up fixing it by adding a shim washer to the mating surface that @Matszwe02 is suggesting to change.
>
> I know [...]

Thanks for your observations. If I understand you correctly, you mention mostly the deformation of that part due to these pretty big forces from the constant tension applied to the filament. As it's really a thing (I had to print it with twice the perimeters and 100% infill to make sure it's as solid as possible), my post mostly mentions the lack of the grab "out of the box", as freshly printed guidler cannot print tpu. (so my change extends the possible creep that won't yet cause extrusion issues) Also I saw it's rubbing on the thread of a bolt that's it's axle, so I think this part is just flawed in many ways.

dr-occams-r, 2024-09-13:
I can confirm both this problem (limited grip) and the deformation problem #340.  I just assumed it was my build or prints the problem since the Voron V0 should just work robustly for printing TPU and also for printing ABS (higher temps) 

Here are several notes that might help other users and/or the project:

-I observed part deformation at regular print temperatures, see #340 possible solution is reducing amps of extruder motor as it is probably adding heat to the part.

-The "anti-squish screw" is documented as required, but I don't seen how the tension could be so high to cause failure of the printing.  However, if the "anti-squish screw" is being pressed against, the part will deform, as mentioned in #340.  So what is the point of it?  Maybe mark is as optional or remove it?  In my case, it only caused problems and no benefit.  Tension should be controlled by the knob and users should be familiar with this.

-I manually sanded the part down (basically the same as the suggested upgraded part), once it worked, I figured others have have had the same problem and found these discussions.  The part definitely needs to be updated to be tighter on the filament, if desired, and also tolerant to slight variations that come with printing the parts.
[image]

-The part should definitely be printed at 100%, as others have mentioned.

@Matszwe02: 
-Is this the same part? Will you be keeping it updated with any further changes you make?
https://www.printables.com/model/848709-voron-v0-0-02-v02-improved-guidler

-Is there any way to make the part beefier as suggested in issue #340? 

-Do you see any problems with making the hole deeper for an M2x16 as suggested in #340?

I will do my next builds / rebuilds with the suggested part and report back.

Matszwe02, 2024-09-13:
@dr-occams-r  thanks for the reply

Yes, that's the same part

I basically think the whole design is flawed, as this axis is constantly experiencing a twisting force.

For now I don't have any plans to modify it, I didn't experience any heat creep so I don't know which part of it needs to be beefier. You can download the step file from printables and post a remix, or suggest the modification. 

I also suggest to print everything 100% solid, with as many perimeters as reasonable (I think I had 5 or higher)

disney7, 2024-12-02:
I just wanted to jump in and say that I have this same issue with flexible filaments. My V02 with the Mini StealthBurner grips regular filament superbly, but I can pull TPU out very easily by hand and the extruder won't feed it reliably at all. No issues with my old V01 that has the pre-StealthBurner setup. I'm going to print and try Matszwe02's modified part and will report back.

Driftrotor, 2025-05-14:
now this is interesting, i had similar issues but traced mine back to the pivot screw in the guidler.
over time the hole would ware and become larger until the filament could not be moved consistently due to the guidler being able to twist off axis.
knowing that my personal issue was with the screw being the pivot and that the threads were causing the ware on the printed part i opted to swap it out for a 3mm pin.
i modified the guidler, extruder plate and front cowing for accept the 3mm pin and this for me solved my issue and had been working for me for over a year now with no signs of ware on the parts.
i'm not 100% against the idea that heat soak and pressure are factors involved here but i printed my modified parts with Voron print settings and am yet to see issues arising from these factors.
i personally recommend that a shoulder bolt be used instead of regular type screws for this particular part (guidler pivot) as the cost will be low and also readily available.