VoronDesign/Voron-0 #357: Require tach fan on all new kits

Carlsans, 2024-08-08:
I don't know if you have any power there but it need to be said. I bought a kit from a company I wont disclose here. I went with a dragon hotend. Has you may know, kit fans are reputed to be the cheapest. After less than a week, my hotend fan failed. The hotend mount melted and the hotend forced by pression from the extruder crashed in the bed. The main issue there is that the mount will melt if the fan fail. For me that's a more than major issue, I left my printer for 30 minutes and here is the result : 
[image]
That's how bad the bed was scratched. The kirigami bed absorbed most of the shock even bending it.
With the way kit sellers are concerned with the fan problem (they are not). I think the best way to make all voron printers safe in that regard would be to propose kits that enforce the use of tach fans. Even the best fan can and will fail at some time.

nemgrea, 2024-08-08:
we cannot force kit sellers to use specific parts, we can look at changing the official sourcing guide to include a tach fan but beyond that we have no power in the curation of 3rd party printer kits.

just to be clear this was a standard dragon hoted NOT a dragon HF hotend correct?

Carlsans, 2024-08-08:
Yes it is the standard flow. The hotend screws just went right throught the abs part.
[image]
the other way to fix this would be to have a mount that is completly made of metal so it won't melt.

Carlsans, 2024-08-08:
maybe the use of oversized washers would have delayed the failure...