VoronDesign/Voron-0 #365: Cam Lock/Extrusion Tophat Screw Offset Inconsistency

SnoWFLakE0s, 2025-04-15:
[image]

[image]

Self explanatory issue. As of the current CAD model, the screw in the extrusion tophat that mates with the cam lock screw is not actually at the 5mm offset as specified in the manual.

hartk1213, 2025-04-15:
There is nothing wrong with that,

If you read the manual and look at the diagram it's measuring from the head of the screw , the head of an m3 BHCS is 1.65mm so the exposed thread of 3.35mm plus the 1.65mm is the 5mm mentioned in the manual

closed 2025-04-15 by SnoWFLakE0s

SnoWFLakE0s, 2025-04-15:
> There is nothing wrong with that,
> 
> If you read the manual and look at the diagram it's measuring from the head of the screw , the head of an m3 BHCS is 1.65mm so the exposed thread of 3.35mm plus the 1.65mm is the 5mm mentioned in the manual

My mistake, bad issue. Disoriented by the lock actually colliding w/ the screw. Will close

SnoWFLakE0s, 2025-04-15:
> > > There is nothing wrong with that,
> > > If you read the manual and look at the diagram it's measuring from the head of the screw , the head of an m3 BHCS is 1.65mm so the exposed thread of 3.35mm plus the 1.65mm is the 5mm mentioned in the manual
> > 
> > 
> > My mistake, bad issue. Disoriented by the lock actually colliding w/ the screw. Will close
> 
> Also most of these little issues most likely won't get fixed until the next revision of the printer, also a lot of these happened on exporting/importing the assembly to and from different CAD packages, nemgrea used SolidWorks for all his designs, while a good majority of the Voron team uses Fusion360
> 
> Those minor things pop up overtime as long as there isn't an issue with the design of the printed parts the hardware alignment isn't a huge issue at this time
> 
> But thank you for pointing these out 👍🏼

I understand--I know firsthand what a decentralized engineering process turns out like. I've been trying to optimize motion parts for subtractive manufacturing, and as a result I'm heavily scrutinizing the model. Great work, thank you.